WillCodeForCats/solaredge-modbus-multi #837: Global Dynamic Power Control Timeout

TheLordVader, 2025-06-11:
### Describe the bug

This issue first occured with the new version 3.1.3

### Expected behavior

Expected the add-on not to return any errors as before.

### Screenshots

[image]
[image]
[image]

[image]

### Diagnostic File

This error originated from a custom integration.

Logger: custom_components.solaredge_modbus_multi
Source: helpers/update_coordinator.py:380
integration: SolarEdge Modbus Multi (documentation, issues)
First occurred: June 10, 2025 at 20:21:59 (2 occurrences)
Last logged: 07:29:00

Unexpected error fetching SolarEdge Coordinator data
Traceback (most recent call last):
  File "/config/custom_components/solaredge_modbus_multi/hub.py", line 1686, in read_modbus_data
    inverter_data = await self.hub.modbus_read_holding_registers(
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        unit=self.inverter_unit_id, address=40113, rcount=2
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/config/custom_components/solaredge_modbus_multi/hub.py", line 528, in modbus_read_holding_registers
    result = await self._client.read_holding_registers(
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        address=self._rr_address, count=self._rr_count, slave=self._rr_unit
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/usr/local/lib/python3.13/site-packages/pymodbus/transaction/transaction.py", line 155, in execute
    raise ModbusIOException(txt)
pymodbus.exceptions.ModbusIOException: Modbus Error: [Input/Output] No response received after 3 retries, continue with next request

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 380, in _async_refresh
    self.data = await self._async_update_data()
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solaredge_modbus_multi/__init__.py", line 252, in _async_update_data
    return await self._refresh_modbus_data_with_retry(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    ...<10 lines>...
    )
    ^
  File "/config/custom_components/solaredge_modbus_multi/__init__.py", line 296, in _refresh_modbus_data_with_retry
    raise ex
  File "/config/custom_components/solaredge_modbus_multi/__init__.py", line 293, in _refresh_modbus_data_with_retry
    return await self._hub.async_refresh_modbus_data()
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solaredge_modbus_multi/hub.py", line 428, in async_refresh_modbus_data
    await inverter.read_modbus_data()
  File "/config/custom_components/solaredge_modbus_multi/hub.py", line 1710, in read_modbus_data
    and "No response recieved after" not in e
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
TypeError: argument of type 'ModbusIOException' is not iterable

### Debug logs

```shell

```

### Home Assistant Version

2025.5.3

### solaredge-modbus-multi Version

3.1.3

### Installation Type

HAOS

### Read the Instructions

- [x] My configuration follows https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Configuration
- [x] I have read the Known Issues https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Known-Issues
- [x] This issue is not about a template error (do not report template errors, read https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Template-Design-Notes)

WillCodeForCats, 2025-06-12:
Please try [Release v3.1.4-pre.1](https://github.com/WillCodeForCats/solaredge-modbus-multi/releases/tag/v3.1.4-pre.1)

TheLordVader, 2025-06-13:
Thanks! I'll try the new version immediately. Please see enclosed diagnostics file.

[config_entry-solaredge_modbus_multi-01JK029XXNY4XYTFQJK149SM5C.json](https://github.com/user-attachments/files/20722258/config_entry-solaredge_modbus_multi-01JK029XXNY4XYTFQJK149SM5C.json)

TheLordVader, 2025-06-19:
It seems that the 3.1.4 pre 1 solved the issue, thanks! 😺

WillCodeForCats, 2025-06-21:
Released as [Release v3.1.4](https://github.com/WillCodeForCats/solaredge-modbus-multi/releases/tag/v3.1.4)

closed 2025-06-21 by WillCodeForCats