WillCodeForCats/solaredge-modbus-multi #859: Integration will not connect

Menz01, 2025-07-10:
### Describe the bug

integration will no longer connect and i see this:

Failed setup, will retry: Modbus error: Modbus Error: [Input/Output] Request cancelled outside pymodbus

### Expected behavior

after a restart of the system i am expecting the integration to connect as it always had been

### Screenshots

[image]

[image]

### Diagnostic File

i don't see a "download diagnostics"

[image]

### Debug logs

```shell
everytime i try and upload i am told its too long. not sure what to trim out that you may need to see
```

### Home Assistant Version

2025.7.1

### solaredge-modbus-multi Version

3.1.4

### Installation Type

HAOS

### Read the Instructions

- [x] My configuration follows https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Configuration
- [x] I have read the Known Issues https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Known-Issues
- [x] This issue is not about a template error (do not report template errors, read https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Template-Design-Notes)

Menz01, 2025-07-10:
hmmmm just disabling "auto detect additional entities" seems to have fixed the issue. i wonder why that is?

Spacerat76, 2025-07-13:
> Seems like there's been firmware pushed out that causes issues with Modbus TCP - my inverter went offline overnight.
> 
> I was on 4.22.44. I am now on 4.23.36, which isn't even listed on the SE Website
> 
> At 06:43 the connection failed and no amount of Integration or Home Assistant restarts would bring it back.
> 
> The integrations failed to connect:
> 
> ```
> 2025-07-13 06:43:54.893 WARNING (MainThread) [pymodbus.logging] Failed to connect [Errno 111] Connect call failed ('10.0.111.75', 1502)                                                                                                    

> 
> Probes showed no response at all from the inverter on Port 1502, I had to physically connect to the inverter from my phone and go into the SE App to diagnose.
> 
> It showed that Modbus TCP was still enabled.
> 
> However, once I had set Modbus TCP to disabled, than then back to Enabled it immediately started accepting connections again - hopefully this will help someone else!

I have the same problem and the same firmware.
But I do not have access to the SE App (only monitoring is possible).

Turning off/on the inverter did not work.

I raised a ticket to the SolarEdge support, maybe other can to the same, if they face the same issue!

cyberkryten, 2025-07-13:
You can access the inverter settings by moving the momentary switch to the P side and then connecting to the Wifi access point that appears.

Then look at the Wifi settings to see the 'Default Gateway' given.  Usually this seems to be 172.16.0.1 so put that address into a web browser on the device that is connected and you'll be able to access the settings from there.

You want Site Communication then Modbus TCP.  Change it to Disabled, wait a minute and change it back to Enabled.

Don't touch anything else! All risks your own, etc etc

joelegg, 2025-07-24:
Mine stopped working today too  i'm going to have to work out a way to automatically disable and enable ethernet on this solar edge inverter. Annoying that solar edge  broke it

jluc2808, 2025-07-30:
got some connection failed too, since the 24/07 (after an upgrade of the inverter)
`2025-07-31 00:28:30.874 WARNING (MainThread) [pymodbus.logging] Failed to connect
2025-07-31 00:38:30.522 WARNING (MainThread) [pymodbus.logging] ERROR: expected transaction 443 but got 442, IGNORING.`

Daverover66, 2025-08-02:
I just wanted to drop in and report that disabling and re-enabling the Modbus TCP setting in the inverter settings > Site Communications (SetApp) worked for me.

I've been using this custom integration for about 3 years now.

I updated my inverter firmware to 4.2.36 in early July.  Home Assistant had been reporting quite frequent modbus disconnnections, but they always cleared and reconnected.  

Until last night. 

 I updated Home Assistant Core to 2025.7.4 yesterday morning and at 2133 last night the modbus connection dropped and wouldn't reconnect.   I found this issue and tried unchecking the "Auto-Detect Additional Entities" box but that made no difference.  I left it foir the night knowing that today was going to b a good solar day and the SE app would suffice for now.

This morning I went into the inverter settings in SetApp > Site Communication, for anyone looking, disabled Modbus TCP, waited 30 seconds and enabled it again. Home Assistant picked up the connection again almost instantly.

I'll continue to monitor, but hopefully we have a temporary fix until the repo owner can implement a fix.

I'm no coder by the way, but if there's anything I can do to help, let me know.  Our home energy use is heavily reliant on having the level of control this custom integration gives us so I wouldn't want to see it fall over completely.

Daverover66, 2025-08-03:
That's the second time in 48 hours my modbus connection has been unable to connect with Home Assistant.  Toggling the Modbus TCP setting off and on again worked, again. I've got my polling frequency set to 30s and a Home Assistant automation that reloads the config every 3 hours. Not sure if any of that is helping, or indeed making things worse.

WillCodeForCats, 2025-08-04:
> I'll continue to monitor, but hopefully we have a temporary fix until the repo owner can implement a fix.
>
> I'm no coder by the way, but if there's anything I can do to help, let me know. Our home [...]

There isn't anything I can think of to do with this, and it's worth noting that I don't personally see this problem on my inverters which are both running 0004.0014.0229 which leads me to believe this is a firmware version dependent issue with the inverter.

It's worth noting that the first Modbus/TCP connection needs to be made within 2 minutes or the inverter will close the port, requiring it to be set off and then on again. It's supposed to then leave the port open after the first connection, but I would recommend anyone seeing connection issues that are fixed by cycling the inverter setting to try a maximum of 1 minute polling time in case it's an firmware bug in recent versions that closes the port when it thinks its inactive.

Daverover66, 2025-08-04:
I think you're probably right in thinking this is a firmware version related issue, particularly since your inverters seem to be unaffected.  I need to do a couple of follow-up things to see if I can help resolve those for those of us who are affected.

1. Raise a ticket with Solaredge to see if they can, or will, investigate and possible offer a solution.
2.  Read up on how Modbus/TCP  works from a Solaredge perspective. 

If you can point me towards any documentation you used when you developed this integration I'd be grateful.  I have the Power Control Protocol as well as several other documents, but I'm wondering if there is a comms specific document?

cyberkryten, 2025-08-04:
The latest docs are https://knowledge-center.solaredge.com/sites/kc/files/sunspec-implementation-technical-note.pdf

There's nothing in the latest update to explain what or why they did to cause any issues.  As has been noted by many, this was introduced after a firmware update, so it is a SolarEdge issue and not something that can be fixed by this integration.  

The problem is on the inverter side as for some reason, it closes the TCP port and stops responding.  Nothing can be done about that externally once the port is non-responsive as physical access is required to force it to re-enable with two methods that are reported to work:

a) Manually putting the inverter into it's config mode (using the switch) and then connecting and disabling/re-enabling Modbus over TCP.

b) Physically removing the ethernet cable

For a) there's no way to do this remotely - physical access is required.

For b) depending on your networking gear, HA skills and programming ability you MAY be able to detect and automate the reset (for those with managed switches, logically doing so by disabling the port, waiting and re-enabling will achieve the same thing), but I suspect few have the capability (or the will) to do so.

Daverover66, 2025-08-04:
> two methods that are reported to work:
>Physically removing the ethernet cable

I'm pretty sure I tried this the first time I lost the connection and it didn't do anything 🤔

However if this works:

> depending on your networking gear, HA skills and programming ability you MAY be able to detect and automate the reset (for those with managed switches, logically doing so by disabling the port, [...]

Then I have both the capability and the will 😁 so thanks for that suggestion.

Menz01, 2025-08-04:
Where is the firmware version listed? I only cause so far (knocking on wood
as I type this) I am not having this issue. I would like to confirm if I am
on the same firmware version as you

WillCodeForCats, 2025-08-04:
FW version is an entity and in the device info.

joelegg, 2025-08-06:
I've seen strange behavior with my battery in the last 24 hours. The battery is just completely not being used and just staying at 58% even though it's set to maximize self-consumption.

cyberkryten, 2025-08-06:
> I've seen strange behavior with my battery in the last 24 hours. The battery is just completely not being used and just staying at 58% even though it's set to maximize self-consumption. Backup is set to 10%

Are any of the parameters being updated like the Status and Logbook in HA?  When mine failed the Status changed to Unknown and Logbook showed B_STATUS_OFF but the SOC constantly reported the same value. 

Have a look at the indicator lights (you have to take the plastic side panel off) or logging into the SE portal and going to the faults page to see if there's anything there.

Might be worth powering off Inverter, breakers and battery for a while, then powering back up to see if that restores it, otherwise contact SE support

joelegg, 2025-08-06:
> Might be worth powering off Inverter, breakers and battery for a while, then powering back up to see if that restores it, otherwise contact SE support

i'm not comfy doing this - isn't there a particlar order? 

i am seeing data in home assisant - no unknowns but the data is wrong. 

checking sstatus of inveter no errors appearing when i logged into the site locally (wifi) but the green light is sflashing on inverter and the battery.

jluc2808, 2025-08-06:
> i'm not comfy doing this - isn't there a particlar order?
>
> i am seeing data in home assisant - no unknowns but the data is wrong.
>
> checking sstatus of inveter no errors appearing when i logged into [...]

data on monitorng.solaredge are wrong when i look at the number (prodution and consume) because they only take the data when the inverter is producing and forget info while the solar production is 0, so all the caculation of % is wrong, otherwise the data show inside the graphic is OK. 
I see that data filled in the data from inverter/energy meter are either not completed , either wrong (according with the wrong calc of monitoring.solaredge) especially imported and exported data 

This append, for me, just after the support apply the new release (3.2537)  - my inverter is a SE5K and the previous release was 3.2251 (with a good return) 

This is exactly the same time where i encounter the problem with brake of connection

joelegg, 2025-08-06:
I've opened a case with solar Red so hopefully they get back to me. My
electricity provider is having an electricity Spike. I'm on variable
pricing. Amber tomorrow so hopefully it's fixed by then.

Daverover66, 2025-08-06:
> Might be worth powering off Inverter, breakers and battery for a while, then powering back up to see if that restores it, otherwise contact SE support

> i'm not comfy doing this - isn't there a particlar order?

There is indeed

[image]

Checking the Vdc is not crucial, but if you wait 5 minutes after moving the inverter switch to 0 it should be safe.

Once the battery is off, you can turn the main AC (and DC if you have them) rotary disconnector for your system to turn off the inverter as well.  Wait a minute or so then revervse the operation. So main AC (and DCif you have them) rotary disconnector to on wait for the inverter to boot up then power on the battery.

Daverover66, 2025-08-06:
> green light flashing.

means the system is connected to AC but not producing. The blue light means comms is okay.

joelegg, 2025-08-08:
Turns out my inverter was updated on 4th Aug. And it has had the issue since then. Waiting on solaredge to fix now . I need to block my Inverter from getting updates.

Daverover66, 2025-08-08:
For anyone still strugglng, I've set my polling to 5s and so far 🤞it has remained connected for a couple of days now.  I see frequent repairs pop up in the Home Assistant settings, but they disappear after 40s at most as the next polling request is allowed.

Daverover66, 2025-08-08:
That’s Modbus TCP’s transaction ID.

Every Modbus TCP request gets a unique incrementing transaction ID so the client can match the reply to the correct request.

If the response comes back with the wrong ID (in this case one less than expected), the client library (pymodbus) assumes it’s stale or mismatched data and discards it.

Usually means the inverter took too long to respond to a request, so by the time the packet arrived the client had already moved on and sent another request.

If the Modbus server in the inverter is occasionally slow or dropping packets, the HA integration will see these mismatches, discard the “late” data, and keep retrying. Enough consecutive mismatches → “No response after 3 retries” → connection closed → port reopens.

This definitely points to an inverter firmware/modbus stack quirk.  I sent a full stack of device logs to SE the last time this happened to me.  So far, I've had nothing further from SE, but for you, it might be worth making your polling frequency more aggressive, eg 5s, to see if that helps.  5s polling gives Home Assistant plenty of time to retry 3 times before the 2 minute response time closes and the modbus port disconnects.

maksyms, 2025-09-03:
I've just started having this issue myself. I haven't manually updated the inverters (2x SE5000, so no SetApp, but old-fashioned clicking buttons), but I did update Home Assistant core to 2025.9.0 from 2025.8.2. `telnet <solaredge_modbus_ip> 1502` works perfectly fine and connects.

maksyms, 2025-09-03:
Albeit I get a slightly different error message, I think:

[image]

WillCodeForCats, 2025-09-03:
ExceptionResponse is not a failure to connect, it did connect and that is a modbus response o whatever we asked the inverter for. Code 2 means Illegal Address, meaning we asked the inverter for something it says it doesn't support.

Turn on debugging and upload the log so I can see where it stops.

maksyms, 2025-09-03:
I tried switching off autodiscovery of meters. Now it disconnects after looking for a third battery (I only have two):

```
2025-09-03 23:20:54.052 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1B2: B_Manufacturer  <class 'str'>
2025-09-03 23:20:54.052 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1B2: B_Model  <class 'str'>
2025-09-03 23:20:54.052 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1B2: B_Version  <class 'str'>
2025-09-03 23:20:54.052 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1B2: B_SerialNumber  <class 'str'>
2025-09-03 23:20:54.052 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1B2: B_Device_Address 0xe <class 'int'>
2025-09-03 23:20:54.052 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1B2: B_RatedEnergy 0xff7fffff <class 'float'>
2025-09-03 23:20:54.052 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1B2: Battery 2 not usable (rating <=0)
2025-09-03 23:20:54.052 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Looking for battery I1B3
2025-09-03 23:20:54.082 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Disconnecting from 192.168.0.86:1502 (clear_client=False).
2025-09-03 23:20:54.083 DEBUG (MainThread) [custom_components.solaredge_modbus_multi] Finished fetching SolarEdge Coordinator data in 0.360 seconds (success: False)

```

maksyms, 2025-09-03:
I now switched off autodiscovery of batteries. Now it disconnects after getting info about the second inverter:

```
2025-09-03 23:22:40.536 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I2: C_SunSpec_DID 0x1<class 'int'>
2025-09-03 23:22:40.536 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I2: C_SunSpec_Length 0x41<class 'int'>
2025-09-03 23:22:40.536 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I2: C_Device_address 0x2<class 'int'>
2025-09-03 23:22:40.536 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I2: C_Manufacturer SolarEdge<class 'str'>
2025-09-03 23:22:40.537 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I2: C_Model SE5000<class 'str'>
2025-09-03 23:22:40.537 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I2: C_Option <class 'str'>
2025-09-03 23:22:40.537 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I2: C_Version 0003.2725<class 'str'>
2025-09-03 23:22:40.537 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I2: C_SerialNumber 731266A7<class 'str'>
2025-09-03 23:22:40.688 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Disconnecting from 192.168.0.86:1502 (clear_client=False).
2025-09-03 23:22:40.690 DEBUG (MainThread) [custom_components.solaredge_modbus_multi] Finished fetching SolarEdge Coordinator data in 0.568 seconds (success: False)

```

joelegg, 2025-09-04:
should we hold off upgrading to version 2.2? thanks

WillCodeForCats, 2025-09-04:
> should we hold off upgrading to version 2.2? thanks

I don't have a 2.2, the newest one is [Release v3.1.7](https://github.com/WillCodeForCats/solaredge-modbus-multi/releases/tag/v3.1.7)

joelegg, 2025-09-04:
anyone solar edge  get updated today (firmware)? My solaredge batteries have stopped being used again they are fully charged and in max self consumption but my system is not using them. very frustrating

joelegg, 2025-09-05:
I can only get my solaredge producing again if I disable modbus. Enabling it stops it from working after a few minutes

WillCodeForCats, 2026-08-29:
How the integration handles these connection issues should be improved (or fixed) as part of the migration to use the `modbus-connection` library. It's still in pre-release, but the latest version as of this writing is [Release v4.0.0-pre.14](https://github.com/WillCodeForCats/solaredge-modbus-multi/releases/tag/v4.0.0-pre.14)

Pip-pip-idk, 2026-09-09:
I’m having this connection issue too and tried cycling the Modbus settings but to no avail. I’m a total newb, can’t code but I’m tech savvy. 

However, I noticed that the IP address of my inverter as it appears on my router settings ends in .85 but my integration setting (which has been working well for 12 months) ends in .84 so I guess it’s looking at the wrong address. When I try to edit the IP address in the integration settings it doesn’t allow a change because “Host and port is already configured in another hub”.

Any suggestions? Do I need to delete the integration and reinstall?

WillCodeForCats, 2026-09-14:
I'm going to close this issue along with merging #1077 because the change to modbus-connection means we're no longer directly using pymodbus.

closed 2026-09-14 by WillCodeForCats