WillCodeForCats/solaredge-modbus-multi #892: "Not Loaded" - integration completely failed

tonywheeler60, 2025-08-22:
### Describe the bug

After years of using this, and all else is fine on my HA server, the integration has failed in the last 3 days with no changes or updates recently applied (before the failure).  In Devices the integration shows "Not Loaded".  My network is fine, the modbus IP is unchanged and I've confirmed modbus is running in SolarEdge SetApp.  I'd like to check the integration setup but cannot find that (sorry I have basic HA skills) and suspect that this is denied me because the service is not running.  Any help would be greatly appreciated.

The logs show 2 errors that contain "modbus":

Logger: homeassistant.setup
Source: setup.py:278
First occurred: 15:27:24 (1 occurrence)
Last logged: 15:27:24

Setup failed for custom integration 'solaredge_modbus_multi': Requirements for solaredge_modbus_multi not found: ['pymodbus>=3.6.6,<3.8'].

and:

Logger: homeassistant.util.package
Source: util/package.py:159
First occurred: 15:27:22 (3 occurrences)
Last logged: 15:27:24

Unable to install package pymodbus>=3.6.6,<3.8: × No solution found when resolving dependencies: ╰─▶ Because you require pymodbus>=3.6.6,<3.8 and pymodbus==3.9.2, we can conclude that your requirements are unsatisfiable.

### Expected behavior

have the integration start and work :) 

### Diagnostic File

the integration appears not to be started.

### Debug logs

```shell
as above
```

### Home Assistant Version

2025.8.3

### solaredge-modbus-multi Version

4.23.36

### Installation Type

HAOS

### Read the Instructions

- [x] My configuration follows https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Configuration
- [x] I have read the Known Issues https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Known-Issues
- [x] This issue is not about a template error (do not report template errors, read https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Template-Design-Notes)

WillCodeForCats, 2025-08-22:
You're running an old version of the integration, you need to upgrade it. Install the latest release.

tonywheeler60, 2025-08-22:
Thanks, but sorry but how do I do that?  Normally I just accept the update offered but there is nothing to show updates.  If I downlolad and install the latest version from HACS would that work and accept all my current configuration?  TIA
Tony

WillCodeForCats, 2025-08-22:
Yes just download the latest version (3.1.6) through HACS.

Someone else reported HACS wasn't offering the update until they manually refreshed it, but I don't know why that would be.

WillCodeForCats, 2025-08-24:
> Hi Seth. I'd love to make a donation to you for this excellent work. I'm using a mobile and it's not obvious how to do so - could you help?

I use GitHub Sponsors: https://github.com/sponsors/WillCodeForCats

closed 2025-08-28 by WillCodeForCats