WillCodeForCats/solaredge-modbus-multi #897: Intergation may not start on HA 2025.9.0

WillCodeForCats, 2025-09-03:
Integration won't start on HA 2025.9.0

Unknown change in pymodbus from 3.9.2 (In HA 2025.8) to 3.11.1 (in HA 2025.9) that isn't documented in the pymodbus API changes file:

https://github.com/pymodbus-dev/pymodbus/blob/1ea0c6f848d2d4c444bf49cb570652ed9ef6077f/API_changes.rst

Will need to figure out what changed and adjust accordingly.

gieljnssns, 2025-09-03:
https://github.com/pymodbus-dev/pymodbus/discussions/2713

maksyms, 2025-09-04:
@WillCodeForCats - I can confirm that Release v3.1.7 works perfectly fine with HA Core 2025.9.0 - thank you!

The way I updated was this:
1. Update Solaredge-Modbus-Multi to 3.1.7 while running HA Core 2025.8.3
2. Without restarting, immediately update to HA Core 2025.9.0 and restart

All works - thanks again! 👍

WillCodeForCats, 2025-09-04:
Everyone can update to v3.1.7 since it will still work 2025.9, 2025.8 through 2025.2.

This release is not locked to 2025.9 or a specific pymodbus version, so if you need to roll back to 2025.8 after upgrading to 2025.9 for some reason it will work either way.

My approach is to avoid locking releases to a certain version of Home Assistant. I personally don't immediately update to the first .0 release so I'm running release v3.1.7 on HA 2025.8.3 for now. At some point I will set a new minimum version since HA does depreciate/improve internal API over time, but I don't want to force people to upgrade their HA immediately to get the latest release if I don't have to.

WillCodeForCats, 2025-09-04:
Yes you need to upgrade to v3.1.7 first, either running it normally on 2025.8 like I am (I will upgrade my personal system to 2025.9.1 when it releases) or installing it and then immediately installing 2025.9.0 like @maksyms said.